screekworkshop/screek-human-sensor #6: Build fails for 2A on ESPHome 2023.09

Pabsilon, 2023-10-04:
I can't update my 2a sensors to the latest esphome version because there's a build error when updating.

starting with: 

> ERROR Unable to import component improv_serial:

There's plenty of errors. They seem to be related to the external components? 
I haven't had an issue up to this point.

screekworkshop, 2023-10-04:
Thanks for the feedback, we'll try to investigate the situation tomorrow.

screekworkshop, 2023-10-11:
I am very sorry, but we may not be able to find the time until a few days afterward. There are some family emergencies and there is no time these days when this can be done.

Pabsilon, 2023-10-12:
But of course, don't worry! Take your time, make sure everyone is all right. :)

nebulous, 2023-10-13:
Each component which doesn't load fails with this error: `ImportError: cannot import name 'VARIANT_ESP32C2' from 'esphome.components.esp32.const'`

tunip, 2023-10-14:
With ESPHome 2023.9 you need to remove the external_components: part in the yaml.

screekworkshop, 2023-10-16:
We hope to be able to update the external libraries section tomorrow before it's too late; one of them modifies the core system libraries, which causes an out-of-sync exception with the new code.

Currently removing external libraries can be done as part of the

nebulous, 2023-11-14:
Is there any remedy for this yet? Is the recommendation to downgrade esphome to a particular version in the mean time?

luigi311, 2023-11-19:
I think this has been fixed now as i no longer get errors when building on 2023.11.2

SLboat, 2023-11-19:
Yes, I'm very sorry we haven't replied to the new situation here, we actually finished updating the dependency library a couple weeks ago.

luigi311, 2023-11-19:
ohh thats great to hear! Sounds like we can go ahead and close this issue then. Thank you very much

screekworkshop, 2023-11-19:
Yes, it was very slow some time ago, and I hope to continue exploring some of the firmware side of things with you all when I get a chance next.  
I'll close this case for now.  

thanks

closed 2023-11-19 by screekworkshop