screekworkshop/screek-human-sensor #9: Disconnecting frequently from WiFi

HoM3r17, 2023-12-13:
Hi,

I built two sensors using your code based on the 1u, and I've got an issue since day one : 

The esp has troubles to connect to wifi, sometimes it does, sometimes I have to reboot it after a while to try again.

Even when connected, it sometimes loses completely the network, and can't connect again unless I restart it manually.

I've tried to change my wifi channel, channel width from 40 to 20, fixed IP on my DHCP server, and splitting my 5 and 2.4 ghz network SSIDs, but it seems to be the same.

Here is the log I could find : 

```
[W][wifi_esp32:458]: Event: Disconnected ssid='Freebox-504602' bssid=[redacted] reason='Beacon Timeout'
[W][wifi:604]: Error while connecting to network.
[W][wifi:640]: Restarting WiFi adapter...
[I][wifi:286]: WiFi Connecting to 'Freebox-504602'...

[...]

[D][wifi:456]: Found networks:
[I][wifi:500]: - 'Freebox-504602' [redacted]▂▄▆█
[D][wifi:501]:     Channel: 6
[D][wifi:502]:     RSSI: -78 dB
[I][wifi:500]: - 'Freebox-504602' [redacted]▂▄▆█
[D][wifi:501]:     Channel: 6
[D][wifi:502]:     RSSI: -59 dB
[D][wifi:505]: - [redacted] [redacted]▂▄▆█
[D][wifi:505]: - [redacted] [redacted]▂▄▆█
[I][wifi:286]: WiFi Connecting to 'Freebox-504602'...
```

Here is a log when it does connect :
```
[I][wifi:500]: - 'Freebox-504602' [redacted]▂▄▆█
[D][wifi:501]:     Channel: 6
[D][wifi:502]:     RSSI: -57 dB
[I][wifi:500]: - 'Freebox-504602' [redacted]▂▄▆█
[D][wifi:501]:     Channel: 6
[D][wifi:502]:     RSSI: -79 dB
[D][wifi:505]: - [redacted] [redacted]▂▄▆█
[D][wifi:505]: - [redacted] [redacted]▂▄▆█
[I][wifi:286]: WiFi Connecting to 'Freebox-504602'...
[D][binary_sensor:036]: 'Moving Target': Sending state OFF
[D][sensor:094]: 'Moving Distance': Sending state 55.00000 cm with 0 decimals of accuracy
[D][sensor:094]: 'Move Energy': Sending state 0.00000 % with 0 decimals of accuracy
[D][sensor:094]: 'Still Distance': Sending state 55.00000 cm with 0 decimals of accuracy
[D][sensor:094]: 'Detection Distance': Sending state 53.00000 cm with 0 decimals of accuracy
[D][ld2410:208]: LD2410 Sun Light: 8%
[I][wifi:573]: WiFi Connected!
[C][wifi:391]:   Local MAC: 84:F7:03:DD:5C:88
[C][wifi:396]:   SSID: [redacted]
[C][wifi:397]:   IP Address: 192.168.1.230
[C][wifi:399]:   BSSID: [redacted]
[C][wifi:400]:   Hostname: 'capteur_wc-dd5c88'
[C][wifi:402]:   Signal strength: -57 dB ▂▄▆█
[C][wifi:406]:   Channel: 6
[C][wifi:407]:   Subnet: 255.255.255.0
[C][wifi:408]:   Gateway: 192.168.1.254
[C][wifi:409]:   DNS1: 192.168.1.22
[C][wifi:410]:   DNS2: 0.0.0.0
[D][wifi:582]: Disabling AP...
[D][binary_sensor:036]: 'Moving Target': Sending state ON
[D][sensor:094]: 'Moving Distance': Sending state 82.00000 cm with 0 decimals of accuracy
[D][sensor:094]: 'Move Energy': Sending state 74.00000 % with 0 decimals of accuracy
[D][sensor:094]: 'Still Distance': Sending state 98.00000 cm with 0 decimals of accuracy
[D][sensor:094]: 'Detection Distance': Sending state 52.00000 cm with 0 decimals of accuracy
[D][ld2410:208]: LD2410 Sun Light: 8%
[C][web_server:123]: Setting up web server...
[C][sntp:028]: Setting up SNTP...
[... 182 lines cut ...]
[C][mqtt.binary_sensor:022]: MQTT Binary Sensor 'Moving Target':
[C][mqtt.binary_sensor:023]:   State Topic: 'capteur_wc/binary_sensor/moving_target/state'
[C][mqtt.binary_sensor:022]: MQTT Binary Sensor 'Still Target':
[C][mqtt.binary_sensor:023]:   State Topic: 'capteur_wc/binary_sensor/still_target/state'
[D][sensor:094]: 'Moving Distance': Sending state 79.00000 cm with 0 decimals of accuracy
[D][sensor:094]: 'Move Energy': Sending state 57.00000 % with 0 decimals of accuracy
[D][sensor:094]: 'Still Distance': Sending state 76.00000 cm with 0 decimals of accuracy
[D][sensor:094]: 'Detection Distance': Sending state 62.00000 cm with 0 decimals of accuracy
[D][ld2410:208]: LD2410 Sun Light: 8%
[D][sensor:094]: 'Sun Light': Sending state 8.00000 % with 0 decimals of accuracy
[W][component:214]: Component ld2410 took a long time for an operation (0.09 s).
[W][component:215]: Components should block for at most 20-30ms.
[C][mqtt.sensor:028]: MQTT Sensor 'Moving Distance':
[C][mqtt.sensor:032]:   State Topic: 'capteur_wc/sensor/moving_distance/state'
[C][mqtt.sensor:028]: MQTT Sensor 'Still Distance':
[C][mqtt.sensor:032]:   State Topic: 'capteur_wc/sensor/still_distance/state'
[C][mqtt.sensor:028]: MQTT Sensor 'Move Energy':
[C][mqtt.sensor:032]:   State Topic: 'capteur_wc/sensor/move_energy/state'
[C][mqtt.sensor:028]: MQTT Sensor 'Still Energy':
[C][mqtt.sensor:032]:   State Topic: 'capteur_wc/sensor/still_energy/state'
[C][mqtt.sensor:028]: MQTT Sensor 'Detection Distance':
[C][mqtt.sensor:032]:   State Topic: 'capteur_wc/sensor/detection_distance/state'
[C][mqtt.sensor:028]: MQTT Sensor 'Sun Light':
[C][mqtt.sensor:032]:   State Topic: 'capteur_wc/sensor/sun_light/state'
```

HoM3r17, 2023-12-14:
After testing, it seems to be better with 20Mhz bandwidth instead of 40 on my 2.4 Ghz network.

It held up all day without disconnecting (almost 20 hours)

[image]

I'll update here if anything changes

screekworkshop, 2023-12-15:
Hi, for this case we would suggest to consider rebooting the router, sometimes this can have unexpected benefits, (20M, 40M HT doesn't seem to have much impact), and overall lolin's board will have a little less RF stability.  
But generally in the idle channel, the whole performance will be excellent up.  
I think this may also be partly due to the current limitations of 2.4g (too many devices).

Also some parts of the clone's s2 mini are poorly selected components and the pcb part is not very reasonable, resulting in poorer signals.

closed 2024-01-02 by screekworkshop