screekworkshop/screek-human-sensor #12: [Feature Request] Movement sensitivity

psylenced, 2024-01-02:
As discussed on forums:

There is just a single feature that would make a big difference, and that is to be able to specify the "size" of the area for direction/position values?

The problem is I sit at a desk in front of the sensor, so every second I'm getting micro-movements (forward, backward, static, forward, static, backward, etc). All of these are a few mm different from previous position.

If I could set a minimum distance (100-300mm), then sitting at the desk would be treated as static until I move my whole body.

Is this possible?

Reply (slboat):

> Regarding your suggestion, it sounds like a nice kind of advanced feature, which is a more abstract and advanced variation of mobile.
> 
> But I'm concerned that if it were to be included in the public firmware, it would cause a lot of confusion and puzzlement for a lot of people.
> 
> And at the same time it seems like a problem that could be set up for automation to try to solve, like nodered's workflow that analyzes multiple changes and then virtualizes a static variable.
> 
> I think it's a great idea, but to implement it would present a lot of challenges and problems.

screekworkshop, 2024-01-02:
Hi!
The 2a native firmware doesn't seem to have static and active recognition, it just treats someone existing as existing.  

The idea is bold lies in recognizing the occurrence of activity, micro-motion recognition.  

But how can it be used to automate something?

MStapelfeldt, 2024-02-11:
If you round up/down the mm value to cm or meter, you could compare this in automation if it changes. For example turn off office light while at the desk and switch it on of you move from 50.3cm to 50.8cm. Just store x/y in number as x/10.
Maybe this helps you reach your goal...