screekworkshop/screek-human-sensor #13: human-sensor-2a-stable-github.yaml can't compile with esp-idf

MStapelfeldt, 2024-03-08:
I wanted to add microWakeWord to a device which runs on human-sensor-2a firmware and noticed there are several compile errors:
```

/config/esphome/human-sensor-2a-stable-github.yaml: In lambda function:
/config/esphome/human-sensor-2a-stable-github.yaml:531:32: error: left operand of comma operator has no effect [-Werror=unused-value]
           if (id(zone1_x_begin).state == 0, id(zone1_x_end).state == 0, id(zone1_y_begin).state == 0, id(zone1_y_end).state == 0){
           ~~~~~~~~~~~~~~~~~~~~~^~~~
/config/esphome/human-sensor-2a-stable-github.yaml:531:57: error: right operand of comma operator has no effect [-Werror=unused-value]
           if (id(zone1_x_begin).state == 0, id(zone1_x_end).state == 0, id(zone1_y_begin).state == 0, id(zone1_y_end).state == 0){
                                      ~~~~~~~~~~~~~~~~~~~^~~~
/config/esphome/human-sensor-2a-stable-github.yaml:531:84: error: right operand of comma operator has no effect [-Werror=unused-value]
           if (id(zone1_x_begin).state == 0, id(zone1_x_end).state == 0, id(zone1_y_begin).state == 0, id(zone1_y_end).state == 0){
                                                               ~~~~~~~~~~~~~~~~~~~~~^~~~
/config/esphome/human-sensor-2a-stable-github.yaml: In lambda function:
/config/esphome/human-sensor-2a-stable-github.yaml:553:32: error: left operand of comma operator has no effect [-Werror=unused-value]
           if (id(zone2_x_begin).state == 0, id(zone2_x_end).state == 0, id(zone2_y_begin).state == 0, id(zone2_y_end).state == 0){
           ~~~~~~~~~~~~~~~~~~~~~^~~~
/config/esphome/human-sensor-2a-stable-github.yaml:553:57: error: right operand of comma operator has no effect [-Werror=unused-value]
           if (id(zone2_x_begin).state == 0, id(zone2_x_end).state == 0, id(zone2_y_begin).state == 0, id(zone2_y_end).state == 0){
                                      ~~~~~~~~~~~~~~~~~~~^~~~
/config/esphome/human-sensor-2a-stable-github.yaml:553:84: error: right operand of comma operator has no effect [-Werror=unused-value]
           if (id(zone2_x_begin).state == 0, id(zone2_x_end).state == 0, id(zone2_y_begin).state == 0, id(zone2_y_end).state == 0){
                                                               ~~~~~~~~~~~~~~~~~~~~~^~~~
/config/esphome/human-sensor-2a-stable-github.yaml: In lambda function:
/config/esphome/human-sensor-2a-stable-github.yaml:575:32: error: left operand of comma operator has no effect [-Werror=unused-value]
           if (id(zone3_x_begin).state == 0, id(zone3_x_end).state == 0, id(zone3_y_begin).state == 0, id(zone3_y_end).state == 0){
           ~~~~~~~~~~~~~~~~~~~~~^~~~
/config/esphome/human-sensor-2a-stable-github.yaml:575:57: error: right operand of comma operator has no effect [-Werror=unused-value]
           if (id(zone3_x_begin).state == 0, id(zone3_x_end).state == 0, id(zone3_y_begin).state == 0, id(zone3_y_end).state == 0){
                                      ~~~~~~~~~~~~~~~~~~~^~~~
/config/esphome/human-sensor-2a-stable-github.yaml:575:84: error: right operand of comma operator has no effect [-Werror=unused-value]
           if (id(zone3_x_begin).state == 0, id(zone3_x_end).state == 0, id(zone3_y_begin).state == 0, id(zone3_y_end).state == 0){
                                                               ~~~~~~~~~~~~~~~~~~~~~^~~~
/config/esphome/human-sensor-2a-stable-github.yaml: In lambda function:
/config/esphome/human-sensor-2a-stable-github.yaml:614:14: error: 'temperatureRead' was not declared in this scope
     unit_of_measurement: °C
              ^~~~~~~~~~~~~~~
Compiling .pioenvs/screek-human-sensor-2a/cbor/tinycbor/src/cborencoder_close_container_checked.o
/config/esphome/human-sensor-2a-stable-github.yaml: In lambda function:
/config/esphome/human-sensor-2a-stable-github.yaml:19:28: error: 'ESP' was not declared in this scope
           id(cpu_speed) = ESP.getCpuFreqMHz();
                            ^~~
```
some stuff like `ESP.getCpuFreqMHz()` or `temperatureRead()` could just be commented out, but the main lambda function and validation of zones is an issue. You can reproduce the issue by changing to
```
esp32:
  board: lolin_c3_mini
  framework:
    type: esp-idf
    version: recommended
```

screekworkshop, 2024-03-10:
Hi, we'll investigate support for the esp-idf framework, which may be a bit more time consuming since we rarely use this framework.
Our bp1 is using the esp-idf framework and it has good bluetooth-wifi concurrency. But in general we will prioritize the arduino framework, which has better compatibility.

nliviu, 2024-05-15:
Replace the comma with the and operand `&&` and it will compile.

MStapelfeldt, 2024-05-15:
Fixing it locally for me is easy, I know. But next one will face same issues so I hope it will be merged into stable soon.

screekworkshop, 2024-05-15:
We will update this issue immediately in the next 24 hours! Thanks for the feedback.

SLboat, 2024-05-16:
> Fixing it locally for me is easy, I know. But next one will face same issues so I hope it will be merged into stable soon.

We Sync the beta's current content to stable, considering they've been running for over 7 months.
and now it should fix the proble now.

https://github.com/screekworkshop/screek-human-sensor/commit/bad4dca45cc533831b33b24eecd22ec0411dd449

screekworkshop, 2025-11-14:
yaml updated, use esp-idf now.

closed 2025-11-14 by screekworkshop