screekworkshop/screek-human-sensor #15: [Feature Request] read zones from radar

TomW1605, 2024-03-25:
hi, i have added an antenna to my radar so i can use bluetooth and their app to set zones because i found that easier than using home assistant. but then i realised the zones are not updated in home assistant. the [other esp home library](https://github.com/tsunglung/esphome-ld2450) has a button to pull the zone info from the radar. it would be great if this could have that too

screekworkshop, 2024-03-25:
Thanks for the feedback, that sounds great, of course, might have to configure an extra antenna for all future 2A's, might help with the range of the bluetooth. Currently the case has a spot reserved for an antenna to be mounted.

TomW1605, 2024-03-25:
yeah, i have a bunch of uFL antennas from a past project so i just stuck one of them in and it works fine.

but even without the antenna it would still be good to be able to pull zones from the radar board so they dont get lost if you have to wipe the esp for some reason.

screekworkshop, 2024-03-25:
The idea of pulling the settings is cool, and synchronizing them back seems cool too. We'll look at tsunglunjg's implementation.

screekworkshop, 2024-03-25:
With two sets of region mechanisms, it may also be necessary to turn off the region function of the ld2450, but it may also be a bit of a nuisance to use it just for extracting coordinates.

 Considering that dealing with nothing more than a clump of coordinates, I think it might be really cool to implement a pencil drawing like marking, and recognizing feature using the esp web interface, but of course that would mean a bunch more programming work.

 I must admit though, the way the numbers are entered really sucks, they are so unintuitive.

TomW1605, 2024-03-25:
its mostly ok if you setup the map thing and have the number entities on the same dashboard so you can update them in real time but still not as good as the drag interface in the app.

screekworkshop, 2024-03-25:
Thanks for the feedback, we'll try to do some research.