screekworkshop/screek-human-sensor #16: how does the Zout1 zone work?

TomW1605, 2024-03-25:
im trying to work out how zones work here but things dont seem to match what other documentation for the radar says. spesificaly around Zout1. all the other documentation and implementations i have looked at have no zones, 3 include zones or 3 exclude zones. none of them have the 3 include + 1 exclude setup. this includes the official app.

so i have 2 questions, how is this being done and is there a way to get the 3 exclude mode working with this setup?

screekworkshop, 2024-04-05:
Hi, from the source code, zout1 is implemented by directly excluding this zone from the logical part of the code, i.e. checking if the coordinate where the target is located is in this zone, and if it is, exclude it, and it won't be involved in any zone1-zone3 zone calculations.

Implementing 3 excluded zones is not difficult, but it can make setting up properties for ha very lengthy.

Sorry about the github push message see you we missed it.

TomW1605, 2024-04-05:
yeah i had a look and that seems to be the case for all of the zones. i was assuming you were using the built in zone feature of the sensor but i think i just assumed that because both happen to be limited to 3

screekworkshop, 2024-04-05:
The current zone feature built into the ld2450 results in no output when not in a zone, due to this we chose to keep the earlier option of implementing a separate standalone zone option. Similarly the built in exclude feature causes no output when inside, it looks like it's dead unless the radar software is turned on.
We expect the official output of always raw data to be given so that coordinate and zone matching information is always available.

TomW1605, 2024-04-06:
yeah, i think the way you have done it makes a lot of sense. way eaiser to setup zones when they are done in code rather than through the software.

closed 2024-04-06 by TomW1605