screekworkshop/screek-human-sensor #17: Only one target detected

ady1972, 2024-03-31:
I'm using your yaml file for make some automation in my kitchen.

I hav ecreated two areas and everything work fine, when i enter in these areas light are turning on based on area.
the problem is that when i move from one rea to another and someone else is in that area, only one target is detected.

How can i solve this? Is there any settings to do?

Thanks a lot and have a nice Easter

screekworkshop, 2024-04-01:
hi, For now, please try switching multiplayer mode with hlkradartool to give it a try, hlk only exposes a few usable methods.

ady1972, 2024-04-01:
Thanks a lot I will try asap.
I hope to solve 😁

screekworkshop, 2024-04-05:
Hopefully good progress is being made and we will close this case for the time being so we can reopen it here when necessary.

closed 2024-04-05 by screekworkshop

ady1972, 2024-04-05:
Hello sorry for delay in answwr.
Tried your suggest but option was already active. Same problem. I will change automation for now hoping that in the future there will be some improvement :)