screekworkshop/screek-human-sensor #18: [Question] Power 1U through AC

sayam93, 2024-04-30:
Hi, I plan to buy a bunch of 1U for my usage. However, I'd like to mount them on the ceiling and connect directly to AC 220-240v power using HLK 2M03 power module from HiLink (https://www.hlktech.net/index.php?id=1158 and data sheet at https://robu.in/wp-content/uploads/2020/02/1912111437_HI-LINK-HLK-2M03_C399249.pdf).

I think I need to connect the 220v AC incoming supply into the two AC points of HLK 2M03 for incoming supply. How do I connect the HLK 2M03 to 1U (S2 Mini with HLK2410C)? Can you guide me on how this can be done?

P.S. I am new to this so any guidance would be appreciated!

OwlBawl, 2024-04-30:
I’d recommend you not to connect it directly to 1U, but just solder / connect usb cable to it, and then just ordinary power it via USB

sayam93, 2024-04-30:
could I not connect the HLK 2M03 directly on the S2 Mini board and skip the usb c altogether? Also, should the input be 3.3V or 5V?

OwlBawl, 2024-04-30:
Can’t understand what you mean

sayam93, 2024-04-30:
I mean can I directly connect a 240V AC to 5V DC power module to the s2 mini board which already has HLK LD2410C soldered onto it? I do not want to use a usb c cable to power this.
[image]
[image]

OwlBawl, 2024-04-30:
Yes you can.

VBUS - 5V 
GND

or 3,3V pin

[image]

screekworkshop, 2024-04-30:
[image]
Yes, you have direct access to VCC, GND and they go straight through to the USB power supply. (https://www.wemos.cc/en/latest/_static/files/sch_s2_mini_v1.0.0.pdf)

For power supply specs, please consider 5v, 1a so there is plenty of headroom.
You also need to consider heat dissipation, one of the problems with the ld2410c is that it gets a lot of heat.

sayam93, 2024-04-30:
> <img alt="image" width="342" src="https://private-user-images.githubusercontent.com/126383355/326788552-34f2b434-03df-4d81-99a0-4f04a955f69f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ0Nzc0MTQsIm5iZiI6MTcxNDQ3NzExNCwicGF0aCI6Ii8xMjYzODMzNTUvMzI2Nzg4NTUyLTM0ZjJiNDM0LTAzZGYtNGQ4MS05OWEwLTRmMDRhOTU1ZjY5Zi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQzMFQxMTM4MzRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xNjhmZmRmOWM4NjNmMTJlNTgxNGQzMDIyNmQ5ZWJlYjY0NGJkYjQyNTAyMDZhNTAyMjZmMzBmZTNjYWY4NmQ1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.zsLllVhiviFK-KsNbp5sLXhLhMB14HGTuBdGXYhvlMg"> Yes, you have direct access to VCC, GND and they go straight through to the USB power supply. (https://www.wemos.cc/en/latest/_static/files/sch_s2_mini_v1.0.0.pdf)
> For power supply specs, please consider 5v, 1a so there is plenty of headroom. You also need to consider heat dissipation, one of the problems with the ld2410c is that it gets a lot of heat.

any suggestions on the heat dissipation?

screekworkshop, 2024-04-30:
It needs to be decided on a case-by-case basis, and our experience has been to maintain the possibility of being able to ventilate.

sayam93, 2024-04-30:
[image]

Is this how connections need to be made?

screekworkshop, 2024-05-01:
> ![Test](https://private-user-images.githubusercontent.com/163408168/326794474-0f23ce78-bfe7-4c7b-8d4f-504079eedc0a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ1MjE4NjQsIm5iZiI6MTcxNDUyMTU2NCwicGF0aCI6Ii8xNjM0MDgxNjgvMzI2Nzk0NDc0LTBmMjNjZTc4LWJmZTctNGM3Yi04ZDRmLTUwNDA3OWVlZGMwYS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQzMFQyMzU5MjRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04NWFkM2U1MjcxZjhiZGJjODhlYWE3MzAyNmI4NTFlMTQ1NThmYTk5MzllNWQzMjhhOWZhZjYxNzBiOGQ4NjY0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.dlJho_rb2Cr5Ub3irQoG5ZeYE6jgFD2qztv8-5x4Oxw)
> 
> Is this how connections need to be made?

that should work.

closed 2024-05-08 by sayam93