screekworkshop/screek-human-sensor #22: 404 Error During Compile

restauffer, 2024-07-16:
I am trying to add the F2 to the Home Assistant WebUI. I add the device as a generic and then replace the resulting YAML with that given in this repository. When I try to compile the code I get the following error: "Error: Downloading Firmware Failed: 404" By adding some comments in the YAML I can determine it cannot load

 url: https://github.com/screekworkshop/custom_components_support_for_screek_1w

screekworkshop, 2024-07-17:
[image]
We tested compiling the current yaml on mac and rpi and didn't encounter a 404 error, is it possible to take a screenshot we'll make sure to see what happens to you?
The official issue looks to be still watching.

restauffer, 2024-07-17:
This is what occurs when attempting to use Install from within the WebUI:

[image]

If you remove or comment out the lines inside the redbox on the following image the problem does not occur.

[image]

Note https://github.com/esphome/issues/issues/5926. Other are experienceing the same prioblem.

By the way your screenshot looks like you are using the Esphome Command Line (CLI) compiler instead of the one with the WebUI inside HA. This may be why you are not seeing the problem

diplix, 2024-07-17:
instead of loading external components remotely, you can also load them locally: just copy the repo into your esphome config folder (i created a subfolder `external_components`).

```
external_components:
  #- source:
  #    type: git
  #    url: https://github.com/screekworkshop/custom_components_support_for_screek_2a
  #    ref: main
  #  components: [esp32, uart]
  - source:
      type: local
      path: external_components/custom_components_support_for_screek_2a/components
    components: [esp32, uart]
```

restauffer, 2024-07-17:
@diplix Thanks for the idea. I considered this but then we would miss any updates to the external GitHub repository. Is there a way to automate regular checks/updates of a local repository?

restauffer, 2024-07-17:
I tried @diplix idea and installed the Git files locally and still got the same 404 error.  So it seems the problem is not with getting the GIT files, it is something within the files.

diplix, 2024-07-17:
> Is there a way to automate regular checks/updates of a local repository?

not that i know of, but following/subscribing the repo on github at least can notify you of any updates.

restauffer, 2024-07-17:
I made another discovery. The "404" error only occurs when installing via the USB cable. If you use a generic yaml to flash the device and get it onto wifi, you can then use the full yaml and install it wirelessly without error.

I have sent several emails with info. Let me summarize what I have learned:

1. The problem is a "404" error when trying to install the device using the ESPHome WebUI within Home Assistant.
2.  The problem is somehow related to the External Components section and in particular, the source
   https://github.com/screekworkshop/custom_components_support_for_screek_2a  Commenting out this section eliminates the "404" error. (But causes other errors to appear as those external components are needed)
3.  Moving the aforementioned GitHub repository to the local PC does not stop the "404" error.
4. This problem occurs whether compiling on a Windows 10 PC or a Raspberry Pi 4
5. This problem does not occur if you use the ESPHome CLI to install the file. (But this does not allow using the Secrets file.)
6. This problem only occurs if installing via a USB cable. If the device is already on wifi you can do the install successfully via wifi.

screekworkshop, 2024-07-17:
I'll try to continue writing tomorrow using the data line method to see what might be going on.It's hard to imagine the correlation this would cause from an initial consideration, since the front part of their process should be the same, pulling remote libraries, caching to a temporary compilation directory, and then finally piecing it together, making firmware, and writing firmware.
The logic for generating the binary is probably very close.

In China sometimes it also causes 404 because of blocked github part of the request, but I think it might not be relevant here.

We used the local version when we burned and manufactured the original firmware ourselves.
Currently one of these external dependency libraries can be phased out in the new version and is unlocking an indicator.Another one can also remove the related dependencies if you don't need the serial port logging.

I'll continue to test the direct page install tomorrow and then share our findings with.
Please understand that I'm a bit behind on the esphome front as I've been debugging zigbee's low-power firmware in my spare time lately.

screekworkshop, 2024-07-19:
We tested this a few times using web-ui installed directly to f2, but it didn't trigger a 404 error either.
as this video show:

https://github.com/user-attachments/assets/dea5546a-8378-486a-a75d-9bae91cc37bd

esphome verison is 2024.07

screekworkshop, 2024-07-19:
install with a fresh clean, and seems not 404 error.
https://github.com/user-attachments/assets/e8275c32-e6eb-46bf-959d-058297b9ba06

restauffer, 2024-07-19:
In the video you are choosing "Plug into the computer running Esphome Dashboard" and therefore I assume you are attaching the device via USB to the server running HA.  I am using "Plug into this computer". It is not the one running HA . It is my desktop Windows 10 PC (I get same results when running in a stand-alone Raspberry PI). 

I can try plugging into the PI running HA tomorrow and test similar to what you have done. I do not have access to it tonight.

restauffer, 2024-07-19:
I have tested with  the F2 attached via USB to the Raspberry Pi running HA and it compiles and installs correctly. So we can now say the problem only exists when trying to compile/install with a USB cable on a computer not directly running HA. For most this will be an adequate workaround. For some, who don't have normal physical access to the HA computer, there would still be a problem.

screekworkshop, 2024-07-21:
> I have tested with the F2 attached via USB to the Raspberry Pi running HA and it compiles and installs correctly. So we can now say the problem only exists when trying to compile/install with a USB [...]

Due to time constraints, we haven't been able to try to do this experiment directly on RPI.
But from our brief judgement so far, this could be some sort of problem with esphome's addon in trying to download the dependency libraries.

For the time being, our recommendation would probably be to place them locally to compile.

screekworkshop, 2024-07-21:
[image]
We got a 404 error when we tried to do a download (not a direct USB burn) on the 2024.07, and I think it looks like we replicated the problem. By seeing the file, it should be there, maybe the problem is in the final renaming, which needs further investigation.

lacrimal, 2024-10-10:
Have esphome 2024.9.2 and error also occurs, esphome is isnstalled as docker container.
Other esphome devices give their binary right away however components from screek gives 404 when downloading to local machine (this miachine is not esphome machine).
Using components downloaded locally not resolve problem - it is still 404.

screekworkshop, 2024-10-16:
> Have esphome 2024.9.2 and error also occurs, esphome is isnstalled as docker container. Other esphome devices give their binary right away however components from screek gives 404 when downloading [...]
This may require further investigation, as we have been quite busy recently.
The current method of extracting the bin firmware is to find a file address in the compilation log, which can be used to extract the file.
[image]

lacrimal, 2024-10-16:
Made fork of external components and then patched them according to my esp32c6 but finally dropped that idea and experimented with frameworks and platformio and components. So there is no espClock, and internal temperature since it is not available out of box in esp-idf, but made to get rid with external components like esp32 and uart.
There is issue with device to be online (it is always offline in esphome however it is reporting data of course) and there is also problem with programming it since after putting code altogether you need manually switch esp32c6 to programming mode.
So this is working example for esp32c6+ld2412 under esphome 2024.9.2
```
esphome:
  name: human-sensor-l12
  name_add_mac_suffix: True
  platformio_options:
    platform: https://github.com/platformio/platform-espressif32/archive/refs/tags/v6.7.0.zip
    board_build.f_cpu: 160000000L
    board_build.f_flash: 80000000L
    #board_build.flash_size: 4MB
    #board_upload.flash_size: 4MB
    build_flags: "-DBOARD_HAS_PSRAM"
    board_build.arduino.memory_type: qio_opi
    board_build.partitions: partitions.csv
  on_boot:
    - priority: 300
      then:
        lambda: |-
          id(screek_version).publish_state("BETA_V240712_2");
logger:
  # level: VERY_VERBOSE
  # level: DEBUG

preferences:
    flash_write_interval: 600s

external_components:
  - source:
      type: git
      url: https://github.com/screekworkshop/custom_components_support_for_screek_l12
      ref: main
    components: [ld2412]
  
esp32:
  board: esp32-c6-devkitc-1
  flash_size: 4MB
  variant: esp32c6
  framework:
    type: esp-idf
    version: 5.2.1
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_4MB: y
      COMPILER_OPTIMIZATION_SIZE: y

globals:
  - id: cpu_speed
    type: int
    restore_value: no
    initial_value: '0'
  - id: last_update_ld2450
    type: unsigned long
    restore_value: no
    initial_value: '0'
  - id: init_zone_publish
    type: bool
    restore_value: no
    initial_value: "false"
  - id: last_illuminance
    type: float
    restore_value: no
    initial_value: "-1"

improv_serial:

debug:
  update_interval: 30s

# Enable Home Assistant API
[... 166 lines cut ...]
    on_press:
      - lambda: |-
          id(ld2412_radar) -> factoryReset();

uart:
  id: uart_bus
  tx_pin: 
    number: GPIO16 # D6 TX
    mode:
      input: true
      pullup: true
  rx_pin: 
    number: GPIO17 #D7 RX
    mode:
      input: true
      pullup: true
  baud_rate: 115200
  parity: NONE
  stop_bits: 1
  data_bits: 8

ld2412:
  id: ld2412_radar

```

restauffer, 2024-10-17:
Have you tried connecting the device to the actual PC running HomeAssistant
via USB and then choosing "Plug into computer running EspHome Dashboard"?
That was one option that worked for me back when I originally started this
thread.

Ricke

lacrimal, 2024-11-05:
Way that worked was either:
- remove components and leave only necessary ` components: [ld2412]` it was easy because mos of changes was patched
- put device in boot mode with EN + BOOT combination

screekworkshop, 2025-02-28:
new yaml should work fine.

closed 2025-02-28 by screekworkshop

restauffer, 2025-02-28:
> new yaml should work fine.

Where can one get a copy of the "new yaml"?