screekworkshop/screek-human-sensor #23: Degree Symbol Not UTF-8 Compliant

restauffer, 2024-07-16:
The degree symbol in the line:

unit_of_measurement: °C

is not UTF-8 complaint and causes a compile to fail.

screekworkshop, 2024-07-17:
This would be something very strange, we use this symbol all the time but have not had a compilation error.
We have compiled locally on the mac platform, and also have also tested rpi5's ha on esphome addon compilation, and have not encountered any reported errors.

We tested the file itself and it is also utf-8 encoded:
```
file human-sensor-f2-stable-github.yaml
human-sensor-f2-stable-github.yaml: Unicode text, UTF-8 text
```

restauffer, 2024-07-17:
I discovered a copy I made converted the file  to ANSI. Putting it back to UTF-8 solved the issue.

closed 2024-07-17 by restauffer

screekworkshop, 2024-07-17:
glad to know that:)