screekworkshop/screek-human-sensor #33: F2 cannot update to ESPHome 2025.2.2

Jason-CHK, 2025-03-19:
Hi, I'm using Home Assistant's ESPHome Builder to update the F2 from 2024.10.0 to 2025.2.2. Here is my yaml:
```
substitutions:
  name: screek-human-sensor-f2
  friendly_name: Screek Human Sensor F2

packages:
  Screek.Human-Sensor-F2: github://screekworkshop/screek-human-sensor/f2/yaml/human-sensor-f2-stable.yaml

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false

api:
  encryption: 
    key: "<key>"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```
And here is the error log when I tried to compile:
```
INFO ESPHome 2025.2.2
INFO Reading configuration /config/esphome/screek-human-sensor-f2.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING Ignoring validation error on pin 13; error: This pin cannot be used on ESP32-C3s and is already used by the SPI/PSRAM interface (function: SPIWP)
Failed config

output.gpio: [source /[data/packages/b6a634c8/f2/yaml/human-sensor-f2-stable.yaml:323](http://data/packages/b6a634c8/f2/yaml/human-sensor-f2-stable.yaml:323)]
  platform: gpio
  id: board_info_ed
  
  This pin cannot be used on ESP32-C3s and is already used by the SPI/PSRAM interface (function: SPIHD).
  pin: GPIO12
```
Could you please take a look? Thanks!

Jason-CHK, 2025-03-19:
Now it has this error:
```

INFO ESPHome 2025.2.2
INFO Reading configuration /config/esphome/screek-human-sensor-f2.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING Ignoring validation error on pin 13; error: This pin cannot be used on ESP32-C3s and is already used by the SPI/PSRAM interface (function: SPIWP)
WARNING Ignoring validation error on pin 12; error: This pin cannot be used on ESP32-C3s and is already used by the SPI/PSRAM interface (function: SPIHD)
WARNING Ignoring validation error on pin 12; error: This pin cannot be used on ESP32-C3s and is already used by the SPI/PSRAM interface (function: SPIHD)
Failed config

output.gpio: [source /config/esphome/screek-human-sensor-f2.yaml:22]
  platform: gpio
  
  ID board_info_ed redefined! Check output->0->id.
  id: board_info_ed
  pin: 
    number: 12
    ignore_pin_validation_error: True
    mode: 
      output: True
      input: False
      open_drain: False
      pullup: False
      pulldown: False
    inverted: False
```

screekworkshop, 2025-03-19:
This may be duplicating components, (it should be done to replace the original part)
We recommend that you proceed directly with the full beta code.
https://github.com/screekworkshop/screek-human-sensor/blob/main/f2/yaml/beta-version/human-sensor-f2-beta.yaml

[image]

Jason-CHK, 2025-03-19:
Thanks, beta code works. Would the stable code be fixed with beta code in the future?

screekworkshop, 2025-03-20:
The Stable code has been fixed, please consider using the beta code, which is currently considered stable and we will make it beta as soon as possible.

closed 2025-03-20 by screekworkshop