screekworkshop/screek-human-sensor #36: F2 won't compile

restauffer, 2025-08-01:
Trying to update F2 Human Sensors from ESPHome Version 2025.63 to 2025.7.4 using Esphome console in Home Assistant. 

Following is the log:

```
INFO ESPHome 2025.7.4
INFO Reading configuration /config/esphome/human-sensor-f2-garage.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING Ignoring validation error on pin 13; error: This pin cannot be used on ESP32-C3s and is already used by the SPI/PSRAM interface (function: SPIWP)
WARNING Ignoring validation error on pin 12; error: This pin cannot be used on ESP32-C3s and is already used by the SPI/PSRAM interface (function: SPIHD)
INFO Generating C++ source...
INFO Compiling app...
Processing human-sensor-f2-garage (board: lolin_c3_mini; framework: arduino; platform: platformio/espressif32@6.3.0)
--------------------------------------------------------------------------------
Tool Manager: Installing https://github.com/espressif/arduino-esp32/releases/download/2.0.9/esp32-2.0.9.zip
INFO Installing https://github.com/espressif/arduino-esp32/releases/download/2.0.9/esp32-2.0.9.zip
Unpacking  [####################################]  100%          
Tool Manager: framework-arduinoespressif32@2.0.9 has been installed!
INFO framework-arduinoespressif32@2.0.9 has been installed!
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - framework-arduinoespressif32 @ 2.0.9 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
Library Manager: Installing Networking
Warning! Could not find the package with 'Networking' requirements for your system 'linux_aarch64'
Dependency Graph
|-- AsyncTCP @ 3.4.5
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer @ 3.7.10
|-- ESP32 Async UDP @ 2.0.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Wire @ 2.0.0
|-- ArduinoJson @ 7.4.2
|-- Improv @ 1.2.4
Compiling .pioenvs/human-sensor-f2-garage/src/esphome/components/api/api_connection.cpp.o
riscv32-esp-elf-g++: error: unrecognized command line option '-std=gnu++20'; did you mean '-std=gnu++2a'?
Compiling .pioenvs/human-sensor-f2-garage/src/esphome/components/api/api_frame_helper.cpp.o
*** [.pioenvs/human-sensor-f2-garage/src/esphome/components/api/api_connection.cpp.o] Error 1
riscv32-esp-elf-g++: error: unrecognized command line option '-std=gnu++20'; did you mean '-std=gnu++2a'?
*** [.pioenvs/human-sensor-f2-garage/src/esphome/components/api/api_frame_helper.cpp.o] Error 1
========================= [FAILED] Took 13.43 seconds =========================
```

If I comment out the framework:, type:, version: and platform_version: line in the YAML, the top two warnings about the Arduino framework are gone and the code will compile and seems to be working OK,

esp32:
  board: lolin_c3_mini
  #framework:
    #type: arduino
    #version: 2.0.9
    #platform_version: 6.3.0

Does the YAML need an update?

restauffer, 2025-08-01:
Update:

I used ChatGPT to research the problem. I told ChatGPT I removed the framework lines and this was the response:

yes, commenting out the framework: section is the right solution in this case. Here's a concise breakdown of what happened and why removing those lines fixed it:

⚠️ Root Problem
The original YAML contains this:

yaml
Copy code
framework:
  type: arduino
  version: 2.0.9
  platform_version: 6.3.0

This pins ESPHome to:

Arduino framework 2.0.9 (which doesn't support C++20),

with platformio/espressif32@6.3.0,

and toolchain-riscv32-esp @ 8.4.0+2021r2-patch5.

That toolchain does not support the -std=gnu++20 flag now used by ESPHome >= 2025.7.0.

So, your build fails with:

vbnet
Copy code
riscv32-esp-elf-g++: error: unrecognized command line option '-std=gnu++20'
✅ Why Removing It Works
When you remove the manual framework: section, ESPHome uses its default toolchain and platform settings—updated in 2025.7.4—to match the new C++20 requirements. This includes:

An updated PlatformIO base (e.g. espressif32@6.5.0 or newer),

A newer Arduino core version (2.0.11 or later),

A C++20-compatible toolchain for RISC-V (ESP32-C3).

These defaults are automatically compatible with ESPHome 2025.7.4+.

✅ Final Recommendation
Unless you're debugging a very specific hardware issue, do not manually pin the framework or platform version. ESPHome is tightly integrated with specific versions, and mismatches like this will increasingly break as features (like C++20) are adopted.

So yes, keep those lines commented out or delete them entirely:

yaml
Copy code
# framework:
#   type: arduino
#   version: 2.0.9
#   platform_version: 6.3.0
🧠 Optional: Clean-up Tip
You also have:

yaml
Copy code
platformio_options:
  board_build.flash_mode: dio
This is fine to leave, but if you're no longer doing low-level tuning, you can remove that too unless you specifically require dio mode for boot stability or flash compatibility.

screekworkshop, 2025-08-01:
Thanks for the feedback, we'll look into the new frame as soon as possible.
Yes, you can eliminate those special little magic tricks if you don't need serial logging.

closed 2025-11-05 by restauffer